learnsyslab/safe-control-gym #17: Bug in observation dimension of 1D quadrotor

Wang4758, 2022-03-23:
Hi there,

There is a bug at [this line](https://github.com/utiasDSL/safe-control-gym/blob/main/safe_control_gym/envs/gym_pybullet_drones/quadrotor.py#L34). The observation dimension should be 2 instead of 6 if `self.quad_type= QuadType.ONE_D`.

JacopoPan, 2022-03-23:
Hi @Wang4758,

can you add the command line you are trying to run and the terminal output?
Just changing `quad_type` might not work if the environment is trying to also load constraints/other parameters that are meant for the 2D experiments.

Wang4758, 2022-03-24:
> Hi @Wang4758,
> 
> can you add the command line you are trying to run and the terminal output? Just changing `quad_type` might not work if the environment is trying to also load constraints/other parameters that are meant for the 2D experiments.

I changed the `quad_type` in `tracking.yaml` and ran `python tracking.py --overrides ./tracking.yaml`.

JacopoPan, 2022-03-24:
@Wang4758 

I see. That is expected because that script (`tracking.py`) is specific to the 2D quad (e.g. it includes the drawing/visualization of a 2D trajectory, etc.). 

The simplest way run a 1D quad simulation, without control, is the `verbose_api.py` example:
- changing `quad_type:` to `1`
- changing `gui:` to `True` 
- commenting/removing everything after `normalized_rl_action_space` in the `quadrotor_config` part of `verbose_api.yaml` 
- commenting/removing all the printouts from `verbose_api.py`

However, you are right in saying there is a problem: at line 111 of `quadrotor.py`:
```
info_mse_metric_state_weight=[1,0,1,0,0,0],
```
This default only works for the (6 states of the) 2D quad, it should take value `info_mse_metric_state_weight=[1,0]` for the 1D case (whose state is only [z, z_dot]). I will fix this with an additional if-else ASAP.

closed 2022-03-24 by JacopoPan