learnsyslab/safe-control-gym #23: Installation error: Macbook air

Abhinav-Muraleedharan, 2022-04-13:
Hi I'm getting the following error while installing safe-control-gym on MacBook air.

      In file included from cddlib/lib-src/cdd.h:17:
      cddlib/lib-src/cddmp.h:30:11: fatal error: 'gmp.h' file not found
       #include "gmp.h"
                ^~~~~~~
      1 error generated.
      error: command '/usr/bin/gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pycddlib
  Running setup.py clean for pycddlib
Successfully built safe-control-gym
Failed to build pycddlib
Installing collected packages: pycddlib, pyasn1-modules, pyaml, protobuf, Pillow, packaging, oauthlib, numpy, munch, kiwisolver, joblib, importlib-resources, importlib-metadata, idna, grpcio, fonttools, filelock, dict-deep, deprecated, cycler, cloudpickle, click, charset-normalizer, cachetools, attrs, async-timeout, absl-py, scipy, requests, redis, pandas, Mosek, matplotlib, markdown, jsonschema, imageio, gym, google-auth, scs, scikit-learn, requests-oauthlib, ray, qdldl, pytope, ecos, scikit-optimize, osqp, gpytorch, google-auth-oauthlib, tensorboard, cvxpy, safe-control-gym
  Running setup.py install for pycddlib ... error
  error: subprocess-exited-with-error
  
  × Running setup.py install for pycddlib did not run successfully.
  │ exit code: 1
  ╰─> [21 lines of output]
      /Users/abhinavmuraleedharan/miniconda3/envs/bdp/lib/python3.8/site-packages/setuptools/installer.py:27: SetuptoolsDeprecationWarning: setuptools.installer is deprecated. Requirements should be satisfied by a PEP 517 installer.
        warnings.warn(
      running install
      /Users/abhinavmuraleedharan/miniconda3/envs/bdp/lib/python3.8/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
        warnings.warn(
      running build
      running build_ext
      skipping 'cdd.c' Cython extension (up-to-date)
      building 'cdd' extension
      creating build
      creating build/temp.macosx-10.9-x86_64-3.8
      creating build/temp.macosx-10.9-x86_64-3.8/cddlib
      creating build/temp.macosx-10.9-x86_64-3.8/cddlib/lib-src
      gcc -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -I/Users/abhinavmuraleedharan/miniconda3/envs/bdp/include -arch x86_64 -I/Users/abhinavmuraleedharan/miniconda3/envs/bdp/include -arch x86_64 -DGMPRATIONAL -Icddlib/lib-src -I/Users/abhinavmuraleedharan/miniconda3/envs/bdp/include/python3.8 -c cdd.c -o build/temp.macosx-10.9-x86_64-3.8/cdd.o
      In file included from cdd.c:601:
      In file included from cddlib/lib-src/cdd.h:17:
      cddlib/lib-src/cddmp.h:30:11: fatal error: 'gmp.h' file not found
       #include "gmp.h"
                ^~~~~~~
      1 error generated.
      error: command '/usr/bin/gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> pycddlib

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
(bdp) abhinavmuraleedharan@Abhinavs-MacBook-Air safe-control-gym %

Abhinav-Muraleedharan, 2022-04-13:
I was able to resolve this issue by installing gmp manually on anaconda. 
conda install -c anaconda gmp

JacopoPan, 2022-04-14:
Thanks @Abhinav-Muraleedharan, I think this was noted by @catgloss as well (who proposed to use `apt-get` instead).
Hopefully it will be fixed as soon as I merge #19

closed 2022-04-14 by JacopoPan