learnsyslab/safe-control-gym #25: Try To understand Tracking.py

waranyoghes, 2022-04-22:
Thanks a lot for developing this awesome repo.
I am new to programming and am trying to learn about how the control algorithm is being executed here. 
My first question is:-
[https://github.com/utiasDSL/safe-control-gym/blob/f5a6b16b05cfa7fa5fa21da59c06c4c905b4cb75/safe_control_gym/utils/registration.py#L159)](url)
whenever I pass the argument --algo lqr --task traj_tracking  in the command line, I get key error for traj_tracking not for lqr. From what I understand the key error should also raise for lqr since the registry.specs is an empty dictionary and key lqr would not be found in it. Please help me understand this.
[image]

My second question is:-
when I type this in the cmd line
$python3 tracking.py --overrides ./tracking.yaml  
this line :- [https://github.com/utiasDSL/safe-control-gym/blob/main/examples/tracking.py#L44 ](url)
should give an key error right?  As the registry.specs is empty.
Also in this line:- [https://github.com/utiasDSL/safe-control-gym/blob/main/examples/tracking.py#L48 ](url)
ctrl does not have any instance variable reference. 

Could someone please provide me an explanation.

JacopoPan, 2022-04-23:
Hi @waranyoghes 

Registration and configuration (under [`safe-control-gym/safe_control_gym/utils`](https://github.com/utiasDSL/safe-control-gym/tree/main/safe_control_gym/utils)) ultimately _just_ serve the purpose of creating, merge, carrying around configuration dictionaries that are created in various way (via `argparse` and multiple, overriding `.yaml` files).

It is certainly certainly useful but not the most _conceptual_ contribution of this repo.

For the first question, I do not know what you are trying to do. Please always include all the commands you run and as much terminal output as possible.

For the second question. I doesn't, right? Why do you think it would?

For the third question, `pid` does have a `self.reference` attribute extracted from the environment's `reset()`.

Script `tracking.py` is meant to be run with the PID controller by command `python3 tracking.py --overrides ./tracking.yaml`, I would start from there and look into the implementation of the controller in [`safe-control-gym/safe_control_gym/controllers/pid`](https://github.com/utiasDSL/safe-control-gym/tree/main/safe_control_gym/controllers/pid)

waranyoghes, 2022-04-26:
Thank you @JacopoPan 
It helped me clarify my doubts. 
I did not see the __init()__ file in the safe_control_gym folder.

waranyoghes, 2022-04-26:
No, I forgot to look into this file while importing the necessary functions:- 
https://github.com/utiasDSL/safe-control-gym/blob/main/safe_control_gym/__init__.py

I have one more doubt. Could you please kindly help me with this.

I am trying to understand the PPO controller. But I am not able find the below file.
from safe_control_gym.envs.env_wrappers.vectorized_env import make_vec_envs
It is used in: https://github.com/utiasDSL/safe-control-gym/blob/f5a6b16b05cfa7fa5fa21da59c06c4c905b4cb75/safe_control_gym/controllers/base_controller.py#L32

if self.training:
            # Training and testing.
            self.env = make_vec_envs(env_func, None, self.rollout_batch_size, self.num_workers, seed)
            self.env = VecRecordEpisodeStatistics(self.env, self.deque_size)

waranyoghes, 2022-04-26:
Thanks a lot.
Okay. Will do that. 
Sorry for asking silly questions and thank you for replying.

JacopoPan, 2022-04-26:
No worries: there ARE such things as silly questions :D but they should still be asked and answered 👍