learnsyslab/safe-control-gym #28: Installation problems

farnammn, 2022-05-16:
Hi,

When I'm installing the package, I get these errors. When I try to install `pycddlib`, I also get similar errors. I have already installed `gmp`, as mentioned in the installation instruction. Do you know what could be the problem, can you please help me?

[image]

[image]

farnammn, 2022-05-16:
Hi again, 

I just wanted to use the environments, and I don't need the agents. I don't really need to install the package. Alternatively, can you give me instructions for using the environments in `safe-control-gym/envs` (i.e. the CartPole and Quadrotor class)?

JacopoPan, 2022-05-16:
Hello @farnammn,

thank you for the interest in the project! The error in the screen captures above does seem related to `gmp`, how did you install it (conda, apt-get)?

For the second point, `examples/verbose_api.py` is an entry point to spin the environments in open loop

Just change
- line 101 in `verbose_api.yaml` to `gui: True`
- line 26 in `verbose_api.py` to `ITERATIONS = int(1000)`

and run
```
python3 verbose_api.py --overrides ./verbose_api.yaml --system quadrotor
```

(You'll need to have `pip3 install -e .` the repo first so if you can't fix your first problem, I suggest to remove the `pytope` dependency from `pyproject.toml`)

JacopoPan, 2022-06-30:
@amdee 

I think there have been a few issues with the built of `pycddlib` wheel since a commit from a couple of weeks ago
https://github.com/mcmtroffaes/pycddlib/issues/55
But hopefully they are resolved (otherwise we will look for alternative implementations)

JacopoPan, 2022-06-30:
You might also want to look at this https://github.com/mcmtroffaes/pycddlib/issues/57

closed 2022-06-30 by JacopoPan

amdee, 2022-07-23:
@JacopoPan Yes, I concur. I completely forgot that I had fixed the issue by reinstalling pycddlib again.