learnsyslab/safe-control-gym #34: Align thrust and pwm limits in quadrotor environment

JakobThumm, 2022-06-07:
Hi,

in the quadrotor environment, is there any particular reason why the maximal thrust does not correspond to the maximum PWM?
If I understand it correctly, the thurst is converted into PWM and then into RPM.
The PWM is clipped to:
https://github.com/utiasDSL/safe-control-gym/blob/523d02df1a5616aeb6f666392dd2b5528f7332fa/safe_control_gym/envs/gym_pybullet_drones/quadrotor_utils.py#L47
Since in RL, the action output is usually also clipped
https://github.com/utiasDSL/safe-control-gym/blob/523d02df1a5616aeb6f666392dd2b5528f7332fa/safe_control_gym/envs/gym_pybullet_drones/quadrotor.py#L557
These two clippings should be the same.

Currently, the maximum thrust is defined as
https://github.com/utiasDSL/safe-control-gym/blob/523d02df1a5616aeb6f666392dd2b5528f7332fa/safe_control_gym/envs/gym_pybullet_drones/base_aviary.py#L141

I would suggest to change that to
```
self.MAX_THRUST = self.KF * 4 * (self.PWM2RPM_SCALE * self.MAX_PWM + self.PWM2RPM_CONST)
```
in the constructor of `quadrotor.py` you would need to adjust that value for the 1D and 2D cases:
```
if self.QUAD_TYPE == QuadType.ONE_D:
    self.MAX_THRUST *= 1/4
elif self.QUAD_TYPE == QuadType.TWO_D:
    self.MAX_THRUST *= 1/2
```

JacopoPan, 2022-06-07:
`self.MAX_THRUST` is legacy code from https://github.com/utiasDSL/gym-pybullet-drones (really just based on the thrust-to-weight ratio and kf parameter in the URDF file of a quadrotor).

I agree that the action space as now defined is redundant (as it includes value that would be clipped by `_preprocess_control`) but I think it should be done purely by changing these lines (and not going through `self.MAX_THRUST` anymore):
https://github.com/utiasDSL/safe-control-gym/blob/523d02df1a5616aeb6f666392dd2b5528f7332fa/safe_control_gym/envs/gym_pybullet_drones/quadrotor.py#L489-L492

JacopoPan, 2022-06-22:
Will address by merging #35 as soon as @adamhall and @Justin-Yuan leave their reviews 👍

closed 2022-06-22 by JacopoPan