learnsyslab/safe-control-gym #93: Should controllers be using `U_GOAL` or `U_EQ`

adamhall, 2022-09-07:
I noticed that many controllers have been using `env.U_GOAL` either as a linearization point or a reference in the cost. `U_GOAL` is computed using the **true** system mass, **not the prior mass**, which is what `symbolic.U_EQ` is for. This means some controllers have been accessing prior information they maybe shouldn't be. Ultimately, if we are assuming the controller doesn't have access to the true mass, then any linearization should be done around what the prior mass should be (`U_EQ`, not `U_GOAL`). In general, choosing the wrong linearization point can have a reasonably large impact on controller performance. 

I have noticed that LQR, iLQR, the RL loss functions, and maybe MPSC have been using `U_GOAL` instead of `U_EQ`.

We should either modify these controllers to use `U_EQ`, or modify the MPC controllers to be using `U_GOAL` to be fair across controllers. I think that using `U_EQ`  better reflects a realistic scenario, where one doesn't actually have access to the true mass, but we should discuss what we ultimately want the environment to provide.

JacopoPan, 2022-09-28:
@adamhall should this be close now that #90 was merged?

adamhall, 2022-09-28:
@JacopoPan no. There is still the question of what the RL should use in its cost functions and normalization.

adamhall, 2022-09-29:
continued in issue #102.

closed 2022-09-29 by adamhall