learnsyslab/safe-control-gym #95: Failed to run create_fig6.sh

merdan-9, 2022-09-13:
Hi, I try to run create_fig6.sh, but it failed.
Could you help me with this problem?
Thanks for your help!
#### error
pybullet build time: May 20 2022 19:44:17
/home/trunk/anaconda3/envs/safe/lib/python3.8/site-packages/gym/spaces/box.py:73: UserWarning: WARN: Box bound precision lowered by casting to float32
  logger.warn(
Traceback (most recent call last):
  File "./gp_mpc_experiment.py", line 197, in <module>
    plot_xz_comparison_diag_constraint(prior_run, run, 1)
  File "./gp_mpc_experiment.py", line 27, in plot_xz_comparison_diag_constraint
    ax = plot_2D_comparison_with_prior(state_inds, prior_run, run, goal, init_ind, dir=dir)
  File "./gp_mpc_experiment.py", line 53, in plot_2D_comparison_with_prior
    horizon_cov = run.state_horizon_cov[init_ind]
IndexError: list index out of range

#### info
@1=vertcat(x, x_dot, z, z_dot, theta, theta_dot), @2=vertcat(T1, T2), (mac(mac((0.5*(@1-Xr)'),Q,zeros(1x6)),(@1-Xr),0)+mac(mac((0.5*(@2-Ur)'),R,zeros(1x2)),(@2-Ur),0))
@1=vertcat(x, x_dot, z, z_dot, theta, theta_dot), @2=vertcat(T1, T2), (mac(mac((0.5*(@1-Xr)'),Q,zeros(1x6)),(@1-Xr),0)+mac(mac((0.5*(@2-Ur)'),R,zeros(1x2)),(@2-Ur),0))
@1=vertcat(x, x_dot, z, z_dot, theta, theta_dot), @2=vertcat(T1, T2), (mac(mac((0.5*(@1-Xr)'),Q,zeros(1x6)),(@1-Xr),0)+mac(mac((0.5*(@2-Ur)'),R,zeros(1x2)),(@2-Ur),0))
@1=vertcat(x, x_dot, z, z_dot, theta, theta_dot), @2=vertcat(T1, T2), (mac(mac((0.5*(@1-Xr)'),Q,zeros(1x6)),(@1-Xr),0)+mac(mac((0.5*(@2-Ur)'),R,zeros(1x2)),(@2-Ur),0))
@1=vertcat(x, x_dot, z, z_dot, theta, theta_dot), @2=vertcat(T1, T2), (mac(mac((0.5*(@1-Xr)'),Q,zeros(1x6)),(@1-Xr),0)+mac(mac((0.5*(@2-Ur)'),R,zeros(1x2)),(@2-Ur),0))
Init State:
[-1.  0.  0.  0. -0.  0.]
iter      objective    inf_pr    inf_du     ||d||  lg(rg) ls
   0   5.504332e+01  1.00e+00  1.00e+01  0.00e+00       -  0

####################   qpOASES  --  QP NO.   1   #####################

    Iter   |    StepLength    |       Info       |   nFX   |   nAC    
 ----------+------------------+------------------+---------+--------- 
       0   |   6.164476e-08   |   REM BND   62   |    85   |     0   
       1   |   2.208861e-11   |   REM BND   60   |    84   |     0   
       2   |   4.431082e-11   |   REM BND   56   |    83   |     0   
       3   |   2.233967e-11   |   REM BND   54   |    82   |     0   
       4   |   4.481644e-11   |   REM BND   50   |    81   |     0   
       5   |   2.259504e-11   |   REM BND   48   |    80   |     0   
       6   |   4.533077e-11   |   REM BND   44   |    79   |     0   
       7   |   2.285482e-11   |   REM BND   42   |    78   |     0   
       8   |   4.585399e-11   |   REM BND   38   |    77   |     0   
       9   |   2.311911e-11   |   REM BND   36   |    76   |     0   
      10   |   4.638633e-11   |   REM BND   32   |    75   |     0   
      11   |   2.338800e-11   |   REM BND   30   |    74   |     0   
      12   |   4.692799e-11   |   REM BND   26   |    73   |     0   
      13   |   2.366162e-11   |   REM BND   24   |    72   |     0   
      14   |   4.747919e-11   |   REM BND   20   |    71   |     0   
      15   |   2.394007e-11   |   REM BND   18   |    70   |     0   
      16   |   4.804016e-11   |   REM BND   14   |    69   |     0   
      17   |   2.422347e-11   |   REM BND   12   |    68   |     0   
      18   |   4.861113e-11   |   REM BND    8   |    67   |     0   
      19   |   2.451194e-11   |   REM BND    6   |    66   |     0   
      20   |   4.919233e-11   |   REM BND    2   |    65   |     0   
      21   |   2.480558e-11   |   REM BND    0   |    64   |     0   
      22   |   9.365151e-09   |   ADD CON    3   |    63   |     1   
      23   |   1.295850e-09   |   ADD CON    5   |    62   |     2   
      24   |   7.426724e-10   |   ADD CON   26   |    61   |     3   
      25   |   1.332632e-09   |   ADD CON   28   |    60   |     4   
      26   |   6.982862e-10   |   ADD CON   49   |    59   |     5   
      27   |   1.369280e-09   |   ADD CON   51   |    58   |     6   
      28   |   6.540761e-10   |   ADD CON   72   |    57   |     7   
      29   |   1.405795e-09   |   ADD CON   74   |    56   |     8   
      30   |   6.100412e-10   |   ADD CON   95   |    55   |     9   
      31   |   1.442178e-09   |   ADD CON   97   |    54   |    10   
      32   |   5.661805e-10   |   ADD CON  118   |    53   |    11   
      33   |   1.478429e-09   |   ADD CON  120   |    52   |    12   
      34   |   5.224929e-10   |   ADD CON  141   |    51   |    13   
      35   |   1.514549e-09   |   ADD CON  143   |    50   |    14   
      36   |   4.789775e-10   |   ADD CON  164   |    49   |    15   
      37   |   1.550540e-09   |   ADD CON  166   |    48   |    16   
      38   |   4.356331e-10   |   ADD CON  187   |    47   |    17   
[... 521 lines cut ...]
Total CPU secs in NLP function evaluations           =      0.988

EXIT: Maximum Number of Iterations Exceeded.
      solver  :   t_proc      (avg)   t_wall      (avg)    n_eval
       nlp_f  | 362.00us (  3.58us) 358.95us (  3.55us)       101
       nlp_g  |  65.91ms (652.59us)  65.98ms (653.29us)       101
    nlp_grad  |   2.78ms (  2.78ms)   2.78ms (  2.78ms)         1
  nlp_grad_f  | 372.00us (  3.65us) 374.08us (  3.67us)       102
  nlp_hess_l  | 695.80ms (  6.96ms) 696.06ms (  6.96ms)       100
   nlp_jac_g  | 228.97ms (  2.24ms) 229.22ms (  2.25ms)       102
       total  |   1.11 s (  1.11 s)   1.11 s (  1.11 s)         1
GP Mean eq Contribution: [-3.88652854e-04  2.45557454e-02  2.85799483e-02  4.55388601e-01
 -1.52168405e-03  5.16830598e-03]
True GP value: [-3.93645639e-04  2.33940861e-02  2.85505519e-02  4.66247972e-01
 -1.40092526e-03  4.32021969e-03]
GP SELECT ACTION TIME: 3.138321131002158
0 -th step.
[0.24181908 0.23957112]
[-1.00030505 -0.01170715  0.04174303  0.80051446 -0.02337966 -0.44847958]
-10.0168287032323
True
{'goal_reached': False, 'mse': 1.9188666249301738, 'constraint_values': array([-9.99694948e-01, -3.40282347e+38, -9.17430266e-02, -3.40282347e+38,
       -1.46015014e+00, -3.40282347e+38, -3.00030505e+00, -3.40282347e+38,
       -1.95825697e+00, -3.40282347e+38, -1.50690947e+00, -3.40282347e+38,
       -2.41819078e-01, -2.39571116e-01,  4.18190776e-02,  3.95711161e-02]), 'constraint_violation': 1}

adamhall, 2022-09-13:
Hi,

Thanks for pointing this out. We've made some changes lately and GPMPC looks like it's a little unstable in this example. Can you tell me what machine you are running this on, and I'm assuming the `main` branch?

merdan-9, 2022-09-14:
Thanks for your timely response!
I am running on Ubuntu 18.04 and created env using conda

Here are the packages and their versions
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
_openmp_mutex             5.1                       1_gnu    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
absl-py                   1.2.0                    pypi_0    pypi
aiosignal                 1.2.0                    pypi_0    pypi
attrs                     22.1.0                   pypi_0    pypi
ca-certificates           2022.07.19           h06a4308_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
cachetools                5.2.0                    pypi_0    pypi
casadi                    3.5.5                    pypi_0    pypi
certifi                   2022.6.15        py38h06a4308_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
charset-normalizer        2.1.1                    pypi_0    pypi
click                     8.0.4                    pypi_0    pypi
cloudpickle               2.2.0                    pypi_0    pypi
cvxpy                     1.2.1                    pypi_0    pypi
cycler                    0.11.0                   pypi_0    pypi
dict-deep                 4.1.2                    pypi_0    pypi
distlib                   0.3.6                    pypi_0    pypi
ecos                      2.0.10                   pypi_0    pypi
filelock                  3.8.0                    pypi_0    pypi
fonttools                 4.37.1                   pypi_0    pypi
frozenlist                1.3.1                    pypi_0    pypi
google-auth               2.11.0                   pypi_0    pypi
google-auth-oauthlib      0.4.6                    pypi_0    pypi
gpytorch                  1.9.0                    pypi_0    pypi
grpcio                    1.43.0                   pypi_0    pypi
gym                       0.21.0                   pypi_0    pypi
idna                      3.3                      pypi_0    pypi
imageio                   2.21.3                   pypi_0    pypi
importlib-metadata        4.12.0                   pypi_0    pypi
importlib-resources       5.9.0                    pypi_0    pypi
joblib                    1.1.0                    pypi_0    pypi
jsonschema                4.16.0                   pypi_0    pypi
kiwisolver                1.4.4                    pypi_0    pypi
ld_impl_linux-64          2.38                 h1181459_1    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
libffi                    3.3                  he6710b0_2    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
libgcc-ng                 11.2.0               h1234567_1    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
libgomp                   11.2.0               h1234567_1    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
libstdcxx-ng              11.2.0               h1234567_1    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
linear-operator           0.1.1                    pypi_0    pypi
markdown                  3.4.1                    pypi_0    pypi
markupsafe                2.1.1                    pypi_0    pypi
matplotlib                3.5.3                    pypi_0    pypi
mosek                     9.3.21                   pypi_0    pypi
msgpack                   1.0.4                    pypi_0    pypi
munch                     2.5.0                    pypi_0    pypi
ncurses                   6.3                  h5eee18b_3    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
numpy                     1.23.3                   pypi_0    pypi
oauthlib                  3.2.1                    pypi_0    pypi
openssl                   1.1.1q               h7f8727e_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
osqp                      0.6.2.post5              pypi_0    pypi
packaging                 21.3                     pypi_0    pypi
pandas                    1.4.4                    pypi_0    pypi
pillow                    9.2.0                    pypi_0    pypi
pip                       22.2.2                   pypi_0    pypi
pkgutil-resolve-name      1.3.10                   pypi_0    pypi
platformdirs              2.5.2                    pypi_0    pypi
protobuf                  3.19.4                   pypi_0    pypi
pyaml                     21.10.1                  pypi_0    pypi
pyasn1                    0.4.8                    pypi_0    pypi
pyasn1-modules            0.2.8                    pypi_0    pypi
pybullet                  3.2.5                    pypi_0    pypi
pycddlib                  2.1.6                    pypi_0    pypi
pyparsing                 3.0.9                    pypi_0    pypi
pyrsistent                0.18.1                   pypi_0    pypi
python                    3.8.10               h12debd9_8    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
python-dateutil           2.8.2                    pypi_0    pypi
[... 7 lines cut ...]
requests-oauthlib         1.3.1                    pypi_0    pypi
rsa                       4.9                      pypi_0    pypi
safe-control-gym          0.5.0                    pypi_0    pypi
scikit-learn              1.1.2                    pypi_0    pypi
scikit-optimize           0.9.0                    pypi_0    pypi
[... cut ...]
virtualenv                20.16.5                  pypi_0    pypi
werkzeug                  2.2.2                    pypi_0    pypi
wheel                     0.37.1             pyhd3eb1b0_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
xz                        5.2.5                h7f8727e_1    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
zipp                      3.8.1                    pypi_0    pypi
zlib                      1.2.12               h5eee18b_3    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main

adamhall, 2022-09-14:
Thanks for that! So far I haven't been able to recreate your issue (I'm on Ubuntu 20.04) and the `create_fig6.sh` runs fine for me on the `main` branch, however, we have seen differences in performance across machines before simply due to torch and different optimizers used by casadi. I'll do some more digging later in the week to see if I can recreate it. Just to confirm a couple of things:
1. you are on the `main` branch, right? 
2. Did you run `create_trained_gp_model.sh` first? This isn't required but could be related to why you are getting the error.
3. Did you change any of the yaml files in `config_overrides`?
4. What python version are you using? Generally, everything is confirmed to work with python 3.8.1

merdan-9, 2022-09-15:
Thanks for your patience!
1. my branch is up to date with 'origin/main'.
2. I ran that script, but it still did not work.
3. I did not modify the yaml files.
4. I create conda env with python 3.8.10, as you mentioned in README.md (https://github.com/utiasDSL/safe-control-gym#option-a-recommended-using-conda)

adamhall, 2022-09-20:
Hey, so far I've tried on a couple of different machines but haven't been able to recreate your issue (so far its working as expected on all the ones I've tried), but I haven't gotten my hands on an Ubuntu 18 machine yet. Just wanted to let you know I'm still working on it.

One thing you could try:
1. Checkout the `main` again to make sure it's up-to-date with origin/main again (in particular make sure all the files in https://github.com/utiasDSL/safe-control-gym/tree/main/experiments/annual_reviews/figure6/trained_gp_model are the same as what's in the repo.
2. Change this line in gp_mpc.py 
https://github.com/utiasDSL/safe-control-gym/blob/5ebaa31581d70c4f2ef0f96d119fd82a0dfeadc2/safe_control_gym/controllers/mpc/gp_mpc.py#L547
to read
                `"ipopt.max_iter": 10000,`
3. Rerun `./create_fig6.sh` again from the figure6 directory (**do not** run `create_trained_gp_model.sh` first).

Basically, the problem you are getting is that gpmpc is finding an infeasible solution. Usually, this has happened for me when the GP is poorly trained. However, it seems like the model files are okay as they have worked well for me on other machines, so we are letting gp_mpc run for more iterations to see if this will help the optimization. It's possible the solver `ipopt` is using some different binary files in the background giving us different optimization performance (but I have to dig into this a little more).

closed 2022-10-17 by JacopoPan