learnsyslab/safe-control-gym #105: Please explain the constraints

MukilSaravanan, 2022-10-11:
Please explain the constraints that need to be satisfied along with maximizing the rewards. What does info['constraint values'] mean?

JacopoPan, 2022-10-11:
Hello @MukilSaravanan 

are is this a question relating the IROS 2022 competition?

The constraints are specified in the YAML files, e.g.

https://github.com/utiasDSL/safe-control-gym/blob/a0e7fa2b14a822f62d97a5cc59a6198317b6645a/competition/level3.yaml#L128-L162

simply means that there are upper and lower bound for the input (identical to the `action_space`) and the x, y, z position of the Crazyflies (bounding it to be within a flyable arena of 6 by 6 by 2.1 meters)

JacopoPan, 2022-10-11:
`info['constraint values']` are the numerical evaluations of those constraints (meaning that any negative value in that entry of the dictionary indicates a violation.

JacopoPan, 2022-10-11:
I'm also tagging @adamhall if you have further implementation-specific questions.

MukilSaravanan, 2022-10-15:
> `info['constraint values']` are the numerical evaluations of those constraints (meaning that any negative value in that entry of the dictionary indicates a violation.

It's actually the other way. ie, if any of those numerical evaluations of constraints is positive, it is said to be violating

JacopoPan, 2022-10-15:
> > `info['constraint values']` are the numerical evaluations of those constraints (meaning that any negative value in that entry of the dictionary indicates a violation.
> 
> It's actually the other way. ie, if any of those numerical evaluations of constraints is positive, it is said to be violating

I'm sure you're right (@adamhall can confirm or refute), I don't remember the sign convention by heart.
The single boolean in `info[constraint_violation]` is intended to avoid any ambiguity.

closed 2022-10-17 by MukilSaravanan