learnsyslab/safe-control-gym #117: Adding pytest to the pytoml

Federico-PizarroBejarano, 2022-12-01:
I think pytest should be on the pytoml since its useful for locally running the tests. However, using `poetry add pytest` isn't working for me, fails due to some error in updating casadi. Not sure if someone with more poetry experience could fix this.

JacopoPan, 2022-12-05:
A few questions:
- what branch are you talking about?
- are there scripts/portions of code in the repo that use `pytest` and demand adding it as a requirement of the package?
- have you tried editing the `.toml` file itself?

More generally, what is the use case that is not currently supported and should be covered?

Federico-PizarroBejarano, 2022-12-05:
- I'm talking about the dev-experiment-class branch, or generally any branch with `pytest` tests. 
- The tests in the code cannot be run without `pytest`. The GitHub hooks will still work, but you cannot run the tests manually without having `pytest`. 
- I certainly could modify the `.toml` file directly but I'm pretty sure thats not how its supposed to be done (since it could cause dependency issues I'm guessing). Idk why my attempt to add it properly isn't working, but I don't know poetry very well. 
- The use case is running tests locally, which isn't possible without `pytest`. I'm not sure everyone who is developing the code is running these tests manually often, but it certainly should become common practice.

amrit110, 2023-02-28:
Hi @Federico-PizarroBejarano not sure if this has been resolved but in your `dev-experiment-class` you can simply add `pytest` as a dependency. You can do this using `poetry add pytest` or `poetry add --group test pytest` if you wish to group your dependencies (https://python-poetry.org/docs/master/managing-dependencies/#dependency-groups), and then do `poetry install` to install `pytest` as part of the virtual environment.

amrit110, 2023-02-28:
@JacopoPan I can submit a PR to resolve the issue with casadi based on what you think. I also think that the poetry.lock file can be added so that it doesn't resolve dependencies dynamically every time and just installs the environment that has been tested to work. As another additional point, we could also release `safe-control-gym` as a package via PyPI. This will increase the useability of what you have developed. Would like to hear your thoughts on that!

Federico-PizarroBejarano, 2023-02-28:
@amrit110 thank you! I have no idea how poetry works so any help is really appreciated. Idk about the other issues, but if you could do a PR to `dev-experiment-class` to fix the pytest and toml file issues that would be great, a lock file sounds good to me but idk about @JacopoPan . Thanks again

JacopoPan, 2023-02-28:
@amrit110 

yes please on the PR to branch `dev-experiment-class` for the poetry points
I also agree with PyPI (it was in the works with the summer interns last year) but I would wait until after the next release/big update on `main`

closed 2023-04-20 by JacopoPan