learnsyslab/safe-control-gym #124: Run demo error.

Yang-Yefeng, 2023-03-16:
I occured an error when I was running the demo in ReadME.md.  
I was running: ./create_fig6.sh 

The error:

Traceback (most recent call last):
  File "./gp_mpc_experiment.py", line 197, in <module>
    plot_xz_comparison_diag_constraint(prior_run, run, 1)
  File "./gp_mpc_experiment.py", line 27, in plot_xz_comparison_diag_constraint
    ax = plot_2D_comparison_with_prior(state_inds, prior_run, run, goal, init_ind, dir=dir)
  File "./gp_mpc_experiment.py", line 53, in plot_2D_comparison_with_prior
    horizon_cov = run.state_horizon_cov[init_ind]
IndexError: list index out of range

JacopoPan, 2023-03-20:
@Yang-Yefeng are you running safe-control-gym in its own dedicated condo environment?
@adamhall is this an error you are familiar with>

Federico-PizarroBejarano, 2024-09-20:
Closing due to staleness