learnsyslab/safe-control-gym #125: Feature Request: migrate from gym to gymnasium

elliottower, 2023-03-17:
Hi, would it be possible for safe-control-gym to be upgraded from gym to gymnasium? Gymnasium is the maintained version of openai gym and is compatible with current RL training libraries ([rllib](https://github.com/ray-project/ray/tree/master/rllib) and [tianshou](https://github.com/thu-ml/tianshou) have already migrated, and [stable-baselines3](https://github.com/DLR-RM/stable-baselines3/blob/e5deeed16efb57c34ccdcb14692439154d970527/docs/guide/install.rst#bleeding-edge-version) will soon).

This repository is currently listed in the gymnasium [third party environments](https://gymnasium.farama.org/environments/third_party_environments/) but we are cleaning the list up to only include maintained gymnasium-compatible repositories.

For information about upgrading and compatibility, see [migration guide](https://gymnasium.farama.org/content/migration-guide/) and [gym compatibility](https://gymnasium.farama.org/content/gym_compatibility/).

JacopoPan, 2023-03-20:
Hello @elliottower 
yes, I will attend to this as soon as we have fully migrated `utiasDSL/gym-pybullet-dones` (which I foresee to be a bit simpler).

jkterry1, 2023-05-26:
@JacopoPan Hey, just wanted to follow up on this :)

JacopoPan, 2023-05-27:
Hi Jordan, the `main` branch's dependencies have all been changed from `gym` to `gymnasium`.
We still need to work on changing the `.step()` API to account for `terminated` and `truncated` returns (it should be easy on the environments' side but will require some due diligence on the controllers/experiments side, can we talk about this with the precommit hooks next week, @Federico-PizarroBejarano?)

Federico-PizarroBejarano, 2023-05-27:
@JacopoPan yes happy to talk about this and do the update this week.

Federico-PizarroBejarano, 2024-09-20:
This was accomplished a while ago

closed 2024-09-20 by Federico-PizarroBejarano