learnsyslab/safe-control-gym #127: Failed to visualize cartpole in tutorial.py when setting gui: True

swuuu, 2023-04-11:
Hi,

I'm trying to run the tutorial.py (i.e., `python3 tutorial.py --algo ppo --task cartpole --overrides ./tutorial_configs/tutorial_ppo_cartpole.yaml --output_dir ./tutorial_models --tag tutorial_results/ppo --thread 1 --seed 22`) with the GUI on. To enable the GUI, I have added `gui: True` under the `task_config` field in the tutorial_ppo_cartpole.yaml file. However, this is the error that I get:
[image]
I also didn't change the cartpole.py file and my PyBullet version is the same as the one in the requirements.txt.

Any help would be appreciated.

Thanks,
Stanley

swuuu, 2023-04-12:
Hi @adamhall,

Thanks for your help! 
I'm using Ubuntu 20.04.4.
My packages are:
[image]
The full output is:
```
(safe) stanleyzqwu@Stanley-Ubuntu:~/Documents/McGill/COMP579/final-project/safe-control-gym/walkthroughs$ python3 tutorial.py --algo ppo --task cartpole --overrides ./tutorial_configs/tutorial_ppo_cartpole.yaml --output_dir ./tutorial_models --tag tutorial_results/ppo --thread 1 --seed 222
pybullet build time: Oct 11 2021 20:59:39
/home/stanleyzqwu/miniconda3/envs/safe/lib/python3.8/site-packages/gym/spaces/box.py:73: UserWarning: WARN: Box bound precision lowered by casting to float32
  logger.warn(
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=AMD
GL_RENDERER=Radeon RX 580 Series (POLARIS10, DRM 3.42.0, 5.15.0-69-generic, LLVM 12.0.0)
GL_VERSION=4.6 (Core Profile) Mesa 21.2.6
GL_SHADING_LANGUAGE_VERSION=4.60
pthread_getconcurrency()=0
Version = 4.6 (Core Profile) Mesa 21.2.6
Vendor = AMD
Renderer = Radeon RX 580 Series (POLARIS10, DRM 3.42.0, 5.15.0-69-generic, LLVM 12.0.0)
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
ven = AMD
@1=vertcat(x, x_dot, theta, theta_dot), (mac(mac((0.5*(@1-Xr)'),Q,zeros(1x4)),(@1-Xr),0)+(((0.5*(U-Ur))*R)*(U-Ur)))
ven = AMD
Traceback (most recent call last):
  File "tutorial.py", line 153, in <module>
    func(config)
  File "tutorial.py", line 35, in train
    control_agent = make(config.algo,
  File "/home/stanleyzqwu/Documents/McGill/COMP579/final-project/safe-control-gym/safe_control_gym/utils/registration.py", line 149, in make
    return registry.make(id, *args, **kwargs)
  File "/home/stanleyzqwu/Documents/McGill/COMP579/final-project/safe-control-gym/safe_control_gym/utils/registration.py", line 102, in make
    obj = spec.make(*args, **kwargs)
  File "/home/stanleyzqwu/Documents/McGill/COMP579/final-project/safe-control-gym/safe_control_gym/utils/registration.py", line 79, in make
    obj = cls(*args, **kwargs)
  File "/home/stanleyzqwu/Documents/McGill/COMP579/final-project/safe-control-gym/safe_control_gym/controllers/ppo/ppo.py", line 48, in __init__
    self.env = make_vec_envs(env_func, None, self.rollout_batch_size, self.num_workers, seed)
  File "/home/stanleyzqwu/Documents/McGill/COMP579/final-project/safe-control-gym/safe_control_gym/envs/env_wrappers/vectorized_env/__init__.py", line 65, in make_vec_envs
    return DummyVecEnv(env_fns)
  File "/home/stanleyzqwu/Documents/McGill/COMP579/final-project/safe-control-gym/safe_control_gym/envs/env_wrappers/vectorized_env/dummy_vec_env.py", line 20, in __init__
    self.envs = [fn() for fn in env_fns]
  File "/home/stanleyzqwu/Documents/McGill/COMP579/final-project/safe-control-gym/safe_control_gym/envs/env_wrappers/vectorized_env/dummy_vec_env.py", line 20, in <listcomp>
    self.envs = [fn() for fn in env_fns]
  File "/home/stanleyzqwu/Documents/McGill/COMP579/final-project/safe-control-gym/safe_control_gym/envs/env_wrappers/vectorized_env/__init__.py", line 33, in _thunk
    env = env_func(seed=e_seed, **env_config)
  File "/home/stanleyzqwu/Documents/McGill/COMP579/final-project/safe-control-gym/safe_control_gym/utils/registration.py", line 149, in make
    return registry.make(id, *args, **kwargs)
  File "/home/stanleyzqwu/Documents/McGill/COMP579/final-project/safe-control-gym/safe_control_gym/utils/registration.py", line 102, in make
    obj = spec.make(*args, **kwargs)
  File "/home/stanleyzqwu/Documents/McGill/COMP579/final-project/safe-control-gym/safe_control_gym/utils/registration.py", line 79, in make
    obj = cls(*args, **kwargs)
  File "/home/stanleyzqwu/Documents/McGill/COMP579/final-project/safe-control-gym/safe_control_gym/envs/gym_control/cartpole.py", line 164, in __init__
    self.PYB_CLIENT = p.connect(p.GUI)
pybullet.error: Only one local in-process GUI/GUI_SERVER connection allowed. Use DIRECT connection mode or start a separate GUI physics server (ExampleBrowser, App_SharedMemoryPhysics_GUI, App_SharedMemoryPhysics_VR) and connect over SHARED_MEMORY, UDP or TCP instead.
numActiveThreads = 0
stopping threads
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
```

Federico-PizarroBejarano, 2024-09-20:
Hi @swuuu, I just realized we never got back to you. I am so sorry about this. If you still have any issues, feel free to open a new issue and I will try to respond as fast as possible. For now, I will close this issue.

closed 2024-09-20 by Federico-PizarroBejarano