learnsyslab/safe-control-gym #142: Building Problem

TaneliLab, 2023-10-25:
1. Fail building because of the gym version, change to 0.22.0 then works
[image]

2. Fail building for poetry,  use "main" instead of "master" then works
requires = ["poetry-core @ git+https://github.com/python-poetry/poetry-core.git@**main**"]
[image]

Federico-PizarroBejarano, 2023-10-26:
Hello @TaneliLab, thank you for your interest in our gym. Are you sure you have pulled the latest version of the code? I see you are using version `0.5.0` of the repo but we are now on version `2.0.0`. In the latest code I believe we have fixed this issues.

1. We are no longer using `gym` but instead using `gymnasium`: https://github.com/utiasDSL/safe-control-gym/blob/83fae93172782f7c6e98063da0b2425d3f741f1b/pyproject.toml#L17
2. I believe we fixed this a while ago, and now it should be using `main` instead of `master` there: https://github.com/utiasDSL/safe-control-gym/blob/83fae93172782f7c6e98063da0b2425d3f741f1b/pyproject.toml#L34

Please let me know if this is the problem!

Federico-PizarroBejarano, 2023-10-27:
No worries! @JacopoPan idk if you plan on updating any issues on that branch or if we just leave it as is?

JacopoPan, 2023-10-28:
Hi @TaneliLab and @Federico-PizarroBejarano, I would hold changing `beta-iros-competition` because it is one of the branches still currently used by @spencerteetaert @Williamwenda for data collection.

I would then want to move the SITL support from `safe-control-gym/beta-iros-competition` to `gym-pybullet-drones/main`.

We should leave this issue open for temporary reference.

Federico-PizarroBejarano, 2024-09-20:
Closing this as I don't think it is reminding us to move the SITL to main.

closed 2024-09-20 by Federico-PizarroBejarano