learnsyslab/safe-control-gym #143: Quadrotor: crazyflie motors corresponding to action dimension.

SapanaChaudhary, 2023-10-31:
Hello! 

The base model for quadrotor is Bitcraze Craziflie 2.0 in the X configuration. 
I was wondering here https://github.com/utiasDSL/safe-control-gym/blob/83fae93172782f7c6e98063da0b2425d3f741f1b/safe_control_gym/envs/gym_pybullet_drones/quadrotor.py#L613
which action dimension corresponds to which motor from the image below? 
[image]

Federico-PizarroBejarano, 2023-11-01:
@JacopoPan @spencerteetaert any idea?

JacopoPan, 2023-11-01:
Yes, the XY positions of the inputs/motors on the quadrotor are described in the URDF, e.g., see in the lines below that `prop0` is at positive 2.8cm from the center of the quad, in the quad frame, then the others in CCW order.

https://github.com/utiasDSL/safe-control-gym/blob/83fae93172782f7c6e98063da0b2425d3f741f1b/safe_control_gym/envs/gym_pybullet_drones/assets/cf2x.urdf#L40C1-L43

SapanaChaudhary, 2023-11-01:
Thank you for the reference to the urdf file. 
If the action dimensions are in the same order as prop_links in the urdf file, then action dimensions look like the figure below, as opposed to the figure I had included earlier. 
[image]
Is this correct?

JacopoPan, 2023-11-02:
Yes, you can also just look at these lines mapping inputs and the forces/torque in Bullet
https://github.com/utiasDSL/safe-control-gym/blob/83fae93172782f7c6e98063da0b2425d3f741f1b/safe_control_gym/envs/gym_pybullet_drones/base_aviary.py#L392-L406

Federico-PizarroBejarano, 2024-09-20:
Closing as it seems to have been resolved.

closed 2024-09-20 by Federico-PizarroBejarano