learnsyslab/safe-control-gym #145: Looking to reproduce a result from IROS 2022 competition paper

SapanaChaudhary, 2024-01-06:
Hello! 

I was hoping to reproduce Fig. 2 (below) from your IROS 2022 competition paper titled [A Remote Sim2real Aerial Competition: Fostering Reproducibility and Solutions’ Diversity in Robotics Challenges](https://arxiv.org/pdf/2308.16743.pdf). 
[image]

I had a few questions regarding the figure above: 
1. Which controller was used to produce the figure - traditional control vs RL? 
2. I looked into the beta-iros-competition branch, but, could only see implementations for ["circle", "square", "figure8"] trajectories. Does this mean if I were to try to reproduce the result above using PPO, I would have to implement that trajectory in https://github.com/utiasDSL/safe-control-gym/blob/beta-iros-competition/safe_control_gym/envs/benchmark_env.py#L491?

Thank you, and looking forward to your reply!

JacopoPan, 2024-01-11:
@SapanaChaudhary those are results comparing the use of the Bitcraze Crazyflie firmware [`cmdFullState()`](https://crazyswarm.readthedocs.io/en/latest/api.html#pycrazyswarm.crazyflie.Crazyflie.cmdFullState) interface in simulation and in real-life, @spencerteetaert might correct me but there should be several examples in folder [`dev-sim2real/`](https://github.com/utiasDSL/safe-control-gym/tree/beta-iros-competition/dev-sim2real)

Federico-PizarroBejarano, 2024-09-20:
Closing due to staleness. If you have any further issues @SapanaChaudhary feel free to let me know, either in a new issue or I can reopen this one.

closed 2024-09-20 by Federico-PizarroBejarano