learnsyslab/safe-control-gym #146: How to use 3D Quadrotor with one of the MPC algorithms?

hedaniel7, 2024-01-23:
In the `examples` folder there are no suitable examples for running a MPC algorithm on a 3D drone.

I tried to create my own following the tutorial as a copy of the PID and run 
```
python3 pid/pid_experiment.py     --algo linear_mpc     --task quadrotor     --overrides         ./pid/config_overrides/quadrotor_3D/quadrotor_3D_tracking.yaml
```

but it doesn't work. Could you please explain this more to me and in the tutorial?

Thank you very much

Daniel

adamhall, 2024-01-23:
Hi Dan,

Thanks or your interest in the repo! To help us help you, can you paste the output you see along with your operating system, a list of your environment (using something like `pip3 list` or `conda list`), and the branch you are working on? 

Best,

Adam

hedaniel7, 2024-01-25:
Hi, 

thank you for your reply. I am working on the main branch

This is the terminal output:

```
(safe) dan@SurfBoard3:~/Projects/safe-control-gym/examples$ python3 pid/pid_experiment.py     --algo linear_mpc     --task quadrotor     --overrides         ./pid/config_overrides/quadrotor_3D/quadrotor_3D_tracking.yaml
pybullet build time: Nov 28 2023 23:45:17
/home/dan/anaconda3/envs/safe/lib/python3.10/site-packages/gymnasium/spaces/box.py:130: UserWarning: WARN: Box bound precision lowered by casting to float32
  gym.logger.warn(f"Box bound precision lowered by casting to {self.dtype}")
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=Intel
GL_RENDERER=Mesa Intel(R) Iris(R) Plus Graphics (ICL GT2)
GL_VERSION=4.6 (Core Profile) Mesa 21.2.6
GL_SHADING_LANGUAGE_VERSION=4.60
pthread_getconcurrency()=0
Version = 4.6 (Core Profile) Mesa 21.2.6
Vendor = Intel
Renderer = Mesa Intel(R) Iris(R) Plus Graphics (ICL GT2)
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
ven = Intel
Workaround for some crash in the Intel OpenGL driver on Linux/Ubuntu
CasADi - 2024-01-25 14:10:12 WARNING("The options 't0', 'tf', 'grid' and 'output_t0' have been deprecated.
The same functionality is provided by providing additional input arguments to the 'integrator' function, in particular:
 * Call integrator(..., t0, tf, options) for a single output time, or
 * Call integrator(..., t0, grid, options) for multiple grid points.
The legacy 'output_t0' option can be emulated by including or excluding 't0' in 'grid'.
Backwards compatibility is provided in this release only.") [.../casadi/core/integrator.cpp:515]
Traceback (most recent call last):
  File "/home/dan/Projects/safe-control-gym/examples/pid/pid_experiment.py", line 132, in <module>
    run()
  File "/home/dan/Projects/safe-control-gym/examples/pid/pid_experiment.py", line 46, in run
    ctrl = make(config.algo,
  File "/home/dan/Projects/safe-control-gym/safe_control_gym/utils/registration.py", line 125, in make
    return registry.make(idx, *args, **kwargs)
  File "/home/dan/Projects/safe-control-gym/safe_control_gym/utils/registration.py", line 88, in make
    obj = spec.make(*args, **kwargs)
  File "/home/dan/Projects/safe-control-gym/safe_control_gym/utils/registration.py", line 69, in make
    obj = cls(*args, **kwargs)
  File "/home/dan/Projects/safe-control-gym/safe_control_gym/controllers/mpc/linear_mpc.py", line 60, in __init__
    super().__init__(
  File "/home/dan/Projects/safe-control-gym/safe_control_gym/controllers/mpc/mpc.py", line 69, in __init__
    self.constraints, self.state_constraints_sym, self.input_constraints_sym = reset_constraints(self.env.constraints.constraints)
AttributeError: 'NoneType' object has no attribute 'constraints'
ven = Intel
Workaround for some crash in the Intel OpenGL driver on Linux/Ubuntu
numActiveThreads = 0
stopping threads
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
[... 107 lines cut ...]
six                       1.16.0                   pypi_0    pypi
sqlite                    3.41.2               h5eee18b_0  
tensorboard               2.15.1                   pypi_0    pypi
tensorboard-data-server   0.7.2                    pypi_0    pypi
termcolor                 1.1.0                    pypi_0    pypi
threadpoolctl             3.2.0                    pypi_0    pypi
tk                        8.6.12               h1ccaba5_0  
tomli                     2.0.1                    pypi_0    pypi
torch                     1.13.1                   pypi_0    pypi
typeguard                 2.13.3                   pypi_0    pypi
typing-extensions         4.9.0                    pypi_0    pypi
tzdata                    2023d                h04d1e81_0  
urllib3                   2.1.0                    pypi_0    pypi
virtualenv                20.25.0                  pypi_0    pypi
werkzeug                  3.0.1                    pypi_0    pypi
wheel                     0.41.2          py310h06a4308_0  
xz                        5.4.5                h5eee18b_0  
zlib                      1.2.13               h5eee18b_0  
```

I work on Ubuntu  20.04

`uname -a
Linux SurfBoard3 6.5.6-surface #3 SMP PREEMPT_DYNAMIC Sun Oct 15 20:52:00 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
`

Federico-PizarroBejarano, 2024-01-25:
@adamhall just added some great example code for MPC to the repo: https://github.com/utiasDSL/safe-control-gym/pull/148. Let us know if you can run the examples!

adamhall, 2024-02-05:
Hi @hedaniel7,
Did this work for you? We will close the issue if yes.

adamhall, 2024-02-13:
Closed for inactivity.

closed 2024-02-13 by adamhall