learnsyslab/safe-control-gym #147: Can we apply MPC control to the 3D quadrotor based on the quaternion model?

hsyfpv, 2024-01-24:


adamhall, 2024-01-24:
Currently, in our CasADi model that MPC has access too provided by the environment, we use Euler angles and roll, pitch, and yaw rates for our state vector for the 3D quadrotor. You can see this here https://github.com/utiasDSL/safe-control-gym/blob/83fae93172782f7c6e98063da0b2425d3f741f1b/safe_control_gym/envs/gym_pybullet_drones/quadrotor.py#L511.

If you want to use quaternions instead, you could generate you're own CasADi model inside your controller. It would also be great to have the option to switch between rotation representations inside the environment (i.e., having a parameter to choose a quaternion, Euler, etc. representation), however, we don't have the bandwidth to tackle this right now. If you do end up making a quaternion representation for the 3D quad, we could work together to make an environment variable to switch between the representations.

adamhall, 2024-02-05:
Hi @hsyfpv,
Can we close this issue?

closed 2024-02-13 by adamhall

hsyfpv, 2024-02-17:
> Hi @hsyfpv, Can we close this issue?

Of course! Thanks!