learnsyslab/safe-control-gym #150: Visualization issue

Hongyu-ZHOU, 2024-02-14:
Hi, I successfully install the main branch and I try the example in the readme. The code runs well but the visualization of the simulator seems weird, I attach here the screen recording when I run those examples: [https://drive.google.com/drive/folders/1_ErgB5kx7XACt88V8eGFM4-CgNVWYVnm?usp=sharing].

Federico-PizarroBejarano, 2024-02-14:
Hello @Hongyu-ZHOU, I see what you mean. The flashing at the beginning of both videos is because the environment is resetting a few times while the GUI is on (something we could perhaps fix to make the visualization nicer, not sure). I see the same behaviour on my on laptop. However, it all happens a lot faster for me, both the initial flashes and the trajectories. My theory is that, for whatever reason, your computer is running the experiments somewhat slower than expected, but its otherwise correct. 

To be honest, while using the gym I almost never use the GUI because it is slower than simply running experiments and collecting the data with the GUI off. Unless you really need the visualization often, my advice would be to simply turn the GUI off (you can put `gui=False` here: https://github.com/utiasDSL/safe-control-gym/blob/7ddb35b0951220d302873bdc8ee7ae86c14d48f2/examples/pid/pid_experiment.py#L16). Let me know if you have other questions, and sorry I can't help more!

Hongyu-ZHOU, 2024-02-14:
@Federico-PizarroBejarano Thanks for your reply. I try also a lambda workstation, it is the same. If the results of those examples are correct, I will just leave the visualization as it is.

Federico-PizarroBejarano, 2024-02-14:
Hmmm weird, I am not sure why it would be so much faster on my computer than yours. @adamhall any ideas? Maybe its because you are using a Mac and I am using Ubuntu 20, but I wouldn't expect much difference.

Hongyu-ZHOU, 2024-02-14:
No worries! I just want to make sure the code runs well and the results are correct :)

Federico-PizarroBejarano, 2024-02-15:
Sounds good, I will close the issue then. Thanks the your interest in our gym!

closed 2024-02-15 by Federico-PizarroBejarano