learnsyslab/safe-control-gym #162: Export models to Onnx

zcase, 2024-09-12:
Been trying to export the ppo models to an onnx model but keep running into errors with how it might be implemented. Have any of you exported the trained models to onnx format?

Federico-PizarroBejarano, 2024-09-18:
Hi @zcase, no I am afraid I haven't, and to my knowledge no one else has. I am not familiar with onnx at all unfortunately. What is the benefit of onnx, and why are you using it?

zcase, 2024-09-18:
@Federico-PizarroBejarano Onnx is a model format that allows you to transfer in between different frameworks like TensorFlow and Torch. I have found it really helpful when transitioning things to c++. Torch has a way to export to onnx models really easily, however when I tried it with the PPO controller models. It looked like there were some errors. I did modify somethings and was able to get an export but it may have broken the rest of the algorithm when doing training.

Federico-PizarroBejarano, 2024-09-18:
Rip I am sorry to hear that, but I hope you get it to work! It seems very cool. If more people are interested, we may try to add support for onnx in the gym at some point then. Good luck!

closed 2024-09-18 by Federico-PizarroBejarano