learnsyslab/safe-control-gym #167: [installation error] 'cddlib/setoper.h' file not found

shuoyang2000, 2024-10-07:
Hi! I encountered the following error while trying to install the package (on macosx-13.2-arm64 with python3.10), could you help me on it? Thanks!

Building wheel for pycddlib (pyproject.toml) ... error
error: subprocess-exited-with-error

× Building wheel for pycddlib (pyproject.toml) did not run successfully.
│ exit code: 1
╰─> [29 lines of output]
running bdist_wheel
running build
running build_py
running egg_info
writing src/pycddlib.egg-info/PKG-INFO
writing dependency_links to src/pycddlib.egg-info/dependency_links.txt
writing requirements to src/pycddlib.egg-info/requires.txt
writing top-level names to src/pycddlib.egg-info/top_level.txt
reading manifest file 'src/pycddlib.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
no previously-included directories found matching 'test'
adding license file 'LICENSE.md'
adding license file 'AUTHORS'
writing manifest file 'src/pycddlib.egg-info/SOURCES.txt'
creating build/lib.macosx-13.2-arm64-cpython-310/cdd
copying src/cdd/init.pyi -> build/lib.macosx-13.2-arm64-cpython-310/cdd
copying src/cdd/gmp.pyi -> build/lib.macosx-13.2-arm64-cpython-310/cdd
copying src/cdd/py.typed -> build/lib.macosx-13.2-arm64-cpython-310/cdd
running build_ext
Compiling cython/_cdd.pyx because it changed.
[1/1] Cythonizing cython/_cdd.pyx
building 'cdd.init' extension
creating build/temp.macosx-13.2-arm64-cpython-310/cython
clang -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Users/user_name/safe-control-gym/new_drone_venv/include -I/Users/user_name/.pyenv/versions/3.10.10/include/python3.10 -c cython/_cdd.c -o build/temp.macosx-13.2-arm64-cpython-310/cython/_cdd.o
cython/_cdd.c:1248:10: fatal error: 'cddlib/setoper.h' file not found
#include "cddlib/setoper.h"
^~~~~~~~~~~~~~~~~~
1 error generated.
error: command '/usr/bin/clang' failed with exit code 1
[end of output]

note: This error originates from a subprocess, and is likely not a problem with pip.
ERROR: Failed building wheel for pycddlib
Successfully built safe-control-gym ecos
Failed to build pycddlib
ERROR: Could not build wheels for pycddlib, which is required to install pyproject.toml-based projects

Federico-PizarroBejarano, 2024-10-07:
Unfortunately I also experienced this issue a few days ago and couldn't find a workaround. It must be due to some updated dependency. I will look into this more today!

shuoyang2000, 2024-10-07:
Hello Federico, thanks for your help. I also opened an issue in the pycddlib repo and [Bob ](https://github.com/bobmyhill) suggested to follow [this page](https://pycddlib.readthedocs.io/en/latest/quickstart.html#installation) with two steps (for mac system):
1. brew install cddlib gmp
2. env "CFLAGS=-I$(brew --prefix)/include -L$(brew --prefix)/lib" pip install pycddlib
and it works for me. 
I am copying that repo issue page [here](https://github.com/mcmtroffaes/pycddlib/issues/73) for your reference.

Federico-PizarroBejarano, 2024-10-07:
Thank you! I unfortunately already tried this and it didn't work. I am using Ubuntu 20.04. Let me know if you find a workaround!

Federico-PizarroBejarano, 2024-10-07:
OK I seem to have a workaround. I simply removed the dependency on pytope which seems to be causing the issue. This dependency is only used in `linear_mpsc.py`. For now you can do this while we find a more permanent solution.

Federico-PizarroBejarano, 2024-10-07:
Seems that pytope is installable in python 3.8 but not 3.10. No idea why. For now will raise an issue with them (although doubt they'll answer), and remove pytope from our dependencies until we can find a fix. Considering how small of a dependency it is, we may just indicate that users need to manually install pytope from source if they wish to use it.

Federico-PizarroBejarano, 2024-10-07:
Opened a PR which fixes this in a more permanent fashion: https://github.com/utiasDSL/safe-control-gym/pull/168. This should solve the issue.

closed 2024-10-07 by Federico-PizarroBejarano