learnsyslab/safe-control-gym #177: No initial guess in linear MPC

tfarger, 2024-11-19:
There currently is no initial guess implemented in LinearMPC. 
In regular nonlinear MPC there is an initial guess computation using either ipopt or a constant LQR gain, that sets the initial values of the solver at the first time step, see
https://github.com/utiasDSL/safe-control-gym/blob/253f25ca3490a6b54eb2d9698cc3ab9f1fd8f4e2/safe_control_gym/controllers/mpc/mpc.py#L340-L343

This is not implemented in the LinearMPC class

I am not sure if the initial guess is needed in the linear MPC. I think the MPC solver should get the same result in the first time step without initial guess. It could be useful for solver performance or something similar.

adamhall, 2024-11-19:
Thanks! @MingxuanChe is this intentional? or should we add the line to LinearMPC as well?

MingxuanChe, 2024-11-19:
Thank you for pointing this out. 

- The recent MPC PR #161, to some extent, focuses on improving the runtime. The warm-starting for linear MPC is not prioritized since it runs sufficiently fast. 
- Also, I was hesitant about whether warm-starting linear MPC with a nonlinear ipopt solution makes sense because the linear MPC will satisfy only the linearized dynamics instead of the nonlinear one. But I guess it will be better than not using it.

In general, we could have the initial guess computation for completeness if you don't find any evidence against it.

closed 2026-02-07 by JacopoPan