learnsyslab/safe-control-gym #179: MPC cost set incorrectly in evaluation?

aghezz1, 2024-12-02:
I believe after this line [L86](https://github.com/utiasDSL/safe-control-gym/blob/253f25ca3490a6b54eb2d9698cc3ab9f1fd8f4e2/safe_control_gym/controllers/mpc/mpc.py#L86), the command to update the cost function of the environment is missing (`self.env.set_cost_function_param(self.Q, self.R)`) as done for the LQR controller [L36](https://github.com/utiasDSL/safe-control-gym/blob/253f25ca3490a6b54eb2d9698cc3ab9f1fd8f4e2/safe_control_gym/controllers/lqr/lqr.py#L36).

In the MPC controller, I can see you update the cost in the method `run` at [L458](https://github.com/utiasDSL/safe-control-gym/blob/253f25ca3490a6b54eb2d9698cc3ab9f1fd8f4e2/safe_control_gym/controllers/mpc/mpc.py#L458) but the method `run` is not called when you execute the related [experiment script](https://github.com/utiasDSL/safe-control-gym/blob/main/examples/mpc/mpc_experiment.py).
Therein, the order of function calls is `experiment.run_evaluation(...) > self._execute_evaluations(...) > self._select_action()`.

Can you check if I am right?
If so I can take care of the edit in a separate PR 😄

Federico-PizarroBejarano, 2024-12-02:
Thanks for pointing this out @aghezz1 ! I think this is actually a big error lol, idk how we didn't have a good way to set the quadratic cost before, or maybe I am misunderstanding something

Federico-PizarroBejarano, 2024-12-02:
Yeah I am also not sure what is a 'fair comparison', which is an issue I run into a lot. For example, often I will not penalize the action magnitudes in the RL reward, but this is often necessary in MPC to get well-formed solutions, immediately breaking the symmetry in their costs. I would suggest that the 'fairness' of the comparison is up the the person who sets up the environments and configs and how they choose to set up the costs of the environments and controllers, and not on the software package itself. For RL, the loss during training and evaluation should be the same right, the RL reward set up by the `rew_state_weight` and `rew_act_weight` in the environment config.

adamhall, 2024-12-02:
I'll look a little deeper on this. Good find @aghezz1! 

For evaluation, you are absolutely correct to question how to fairly compare controllers if the function they are optimizing are different than the cost you are evaluating. As @Federico-PizarroBejarano mentioned, it is something we have debated internally for years. The challenge comes in the fact that some controllers/RL simply cannot optimize the same cost function---they are designed to work with a specific form of cost function. 

For example, say you want to have your drone follow a path. You only care about where it goes, not about its rates or orientation or anything else. A cost function that makes sense is to just have a quadratic cost on the position error. RL handles this fine. However, in MPC your R matrix needs to be positive definite. Otherwise you can get some nasty numerical issues. Furthermore, if you want to use LQR, you **need** a p.d. R matrix. In RL, you can just use the quadratic position error (or some other error) and get excellent tracking performance. In fact, we saw that when we included an R matrix in the RL algos, it really degraded their training and performance. So at the end of the day, we decided that the environment defines the evaluation metric that you care about, and the controllers use this cost function if they can, or accordingly and try to perform well on the environment metric. 

I realize that setting `rew_state_weight` might align the cost functions of the algos more, but I don't think this is fair. If your task cares **only** about the position and RL can handle this, then it should be reflected in the cost function for RL. If MPC can handle that same cost function, then great! If it can't, I would say that's a limitation of MPC. I'm open to opinions though. I don't think we've truly found a good solution yet. Another high-level question we grapple with is how to justify data efficiency, data collection, and the impacts of prior models/algorithm structure.

I do think we should remove the `set_cost_function_param`. I'm not entirely sure why that is there?

aghezz1, 2024-12-02:
Thanks for the replies.
I believe that for now I would be happy if when I evaluate the MPC performance I use the same quadratic stage cost that is declared in the MPC problem.
At the moment, in the evaluation script the Q and R matrices are by default two diagonal matrices with only ones and it's not possible to modify them (besides very hacky ways).
I don't know what is the best way to do it within your framework, whether adding a field in yaml of the task configuration or something else

Federico-PizarroBejarano, 2024-12-02:
@aghezz1 I will make a PR right now to fix this and hopefully we can agree on the implementation and get it merged

aghezz1, 2024-12-03:
Thank you for the quick replies! 
I believe you can close this issue and we move the fine-tuning discussion over #181 ?

Federico-PizarroBejarano, 2024-12-03:
Sounds good, no worries! Thanks for raising the issue!

closed 2024-12-03 by Federico-PizarroBejarano