learnsyslab/safe-control-gym #184: Poetry Core Installation Syntax Error

dsdy120, 2025-01-20:
While attempting the following command:
```
pip install -e .
```
as part of the setup procedure (all steps followed exactly on Ubuntu 20.04.6 LTS, python version 3.8.20, miniconda 3.8), the following error message is consistently shown, seems to be part of the poetry-core installation. Requesting advice on how to proceed.
```
Obtaining file:///home/dsdy120/safe-control-gym
  Installing build dependencies ... error
  error: subprocess-exited-with-error
  
  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [44 lines of output]
      Collecting poetry-core@ git+https://github.com/python-poetry/poetry-core.git@main
        Cloning https://github.com/python-poetry/poetry-core.git (to revision main) to /tmp/pip-install-dvge5_5x/poetry-core_573e947ef4f94f43aade586e6dde6dd5
        Running command git clone --filter=blob:none --quiet https://github.com/python-poetry/poetry-core.git /tmp/pip-install-dvge5_5x/poetry-core_573e947ef4f94f43aade586e6dde6dd5
        Resolved https://github.com/python-poetry/poetry-core.git to commit d97577e664d59173482a78f70e8f743cc5468851
        Getting requirements to build wheel: started
        Getting requirements to build wheel: finished with status 'error'
        error: subprocess-exited-with-error
      
        × Getting requirements to build wheel did not run successfully.
        │ exit code: 1
        ╰─> [23 lines of output]
            Traceback (most recent call last):
              File "/home/dsdy120/miniconda3/envs/aer1217-project/lib/python3.8/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
                main()
              File "/home/dsdy120/miniconda3/envs/aer1217-project/lib/python3.8/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
                json_out['return_val'] = hook(**hook_input['kwargs'])
              File "/home/dsdy120/miniconda3/envs/aer1217-project/lib/python3.8/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 112, in get_requires_for_build_wheel
                backend = _build_backend()
              File "/home/dsdy120/miniconda3/envs/aer1217-project/lib/python3.8/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 77, in _build_backend
                obj = import_module(mod_path)
              File "/home/dsdy120/miniconda3/envs/aer1217-project/lib/python3.8/importlib/__init__.py", line 127, in import_module
                return _bootstrap._gcd_import(name[level:], package, level)
              File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
              File "<frozen importlib._bootstrap>", line 991, in _find_and_load
              File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
              File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
              File "<frozen importlib._bootstrap_external>", line 843, in exec_module
              File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
              File "/tmp/pip-install-dvge5_5x/poetry-core_573e947ef4f94f43aade586e6dde6dd5/src/poetry/core/masonry/api.py", line 14, in <module>
                from poetry.core.masonry.builders.wheel import WheelBuilder
              File "/tmp/pip-install-dvge5_5x/poetry-core_573e947ef4f94f43aade586e6dde6dd5/src/poetry/core/masonry/builders/wheel.py", line 122
                os.fdopen(fd, "w+b") as fd_file,
                                     ^
            SyntaxError: invalid syntax
            [end of output]
      
        note: This error originates from a subprocess, and is likely not a problem with pip.
      error: subprocess-exited-with-error
      
      × Getting requirements to build wheel did not run successfully.
      │ exit code: 1
      ╰─> See above for output.
      
      note: This error originates from a subprocess, and is likely not a problem with pip.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

Federico-PizarroBejarano, 2025-01-20:
Hi @dsdy120 , thanks for using our gym! I looked into the issue and it seems poetry no longer supports Python 3.8. It now requires 3.9 and above. I think reasonably this branch of the gym should be updated to Python 3.10 like the main branch, but I am unsure if we are continuing to maintain this branch. I will consult with the team and let you know. Thank you for bringing this to our attention!

Federico-PizarroBejarano, 2025-01-20:
@dsdy120 I found a simple workaround that maybe will be pushed to the branch. Simply replace https://github.com/utiasDSL/safe-control-gym/blob/dc044522d5e6ef7bfbc8cb088862fc8f05f1ad2e/pyproject.toml#L34 with `requires = ["poetry-core==1.9.1"]`

dsdy120, 2025-01-20:
Hi Frederico, thanks for the quick reply! For this workaround should I maintain Python 3.8 or go with Python 3.10 now?

Federico-PizarroBejarano, 2025-01-20:
No worries! For the workaround you can keep using Python 3.8

dsdy120, 2025-01-20:
Hi @Federico-PizarroBejarano , the workaround you presented works well, thank you for the help!

Federico-PizarroBejarano, 2025-01-20:
Awesome! Let me know if you encounter any more issues. I will leave the issue open for the time being until we determine what the plan is for the branch. If I may ask, are you taking AER1217 at UofT or are you using this branch for another reason? I am wondering how many other people are still using this branch

dsdy120, 2025-01-21:
Hi @Federico-PizarroBejarano , the branch is in active use by the AER1217 course which I'm a part of. Not sure if all ~20 students use it over main, but the course instructions we have been given are for this branch so I am currently following it. Thanks again for the help!

Federico-PizarroBejarano, 2025-01-21:
In that case, I will push the fix to the branch and close this issue. The question of what to do with this branch long-term can be discussed separately. Thank you so much! Let me know if you have any further issues.

closed 2025-01-21 by Federico-PizarroBejarano