learnsyslab/safe-control-gym #186: How to visualize in real-time

ssaigarimella, 2025-02-02:
Hi, I am currently using the Cartpole Stabilization with LQR example. I set the gui=True argument within the run() function in lqr_experiment.py, So I am able to visualize the cartpole stabilization via the PyBullet window:

[image]

However, I am unable to visualize it in real-time. The simulation run quickly moves the cartpole but I want to see this in "wall clock time" in my PyBullet window. How can I do this?

Thanks for the great work!

Federico-PizarroBejarano, 2025-02-02:
Hello @ssaigarimella, I have been meaning to find a good solution to this, but a simple solution for now is for you to put a `time.sleep(x)` command in the `select_action(...)` function of LQR (in `lqr.py`). Simply slow down the execution such that it is real time, maxing `x` be equal to the inverse of the control frequency. I am certain there is a better solution, I'll try to look into it on Monday, but for now this works.

ssaigarimella, 2025-02-02:
@Federico-PizarroBejarano the `time.sleep(x) `works perfectly, thanks!

closed 2025-02-02 by ssaigarimella