learnsyslab/safe-control-gym #188: Does it support multi-drone simulation?

Yuan-Mingkang, 2025-02-27:
Hi, I would like to ask if this project supports multiple drone simulations in the same environment. It seems to focus mainly on single drone simulations, is it capable of making modifications for multi-drone simulations?
If so, where would you suggest I start?

Federico-PizarroBejarano, 2025-02-27:
Hi @Yuan-Mingkang, honestly I am unsure how much work it would be to make the gym work for multiple drones. The gym was based on https://github.com/utiasDSL/gym-pybullet-drones, which is specifically made for multiple drones. `gym-pybullet-drones` may be more suitable for your research, and it should be possible to move code you need from our gym to `gym-pybullet-drones`. However, in theory I think it should be possible to modify our gym for this, just cannot comment on how much work it would be.

Yuan-Mingkang, 2025-03-11:
Thank you for your reply~ I have worked on modifying your project and am now able to implement multi-drone simulation. I'm using MPC to control the drones, but I'm running into CPU arithmetic issues, no matter how many drones I run, I can only use 2 cores of the CPU, and they all reach 100% load. This situation is causing my simulation to be very choppy, I would like to use multiprocessing, but after my evaluation, the code modifications are very large, do you have any clue? Make the simulation can operate on more CPU cores.

Federico-PizarroBejarano, 2025-03-11:
Hi @Yuan-Mingkang, I'm happy you've got it working so quickly! Unfortunately I am not sure how to add multiprocessing, I am definitely no expert there. I know that pybullet and MPC in general are CPU intensive and somewhat hard to multi-process. Only thing I can think of is forking a new process to handle calculating the next action for each drone, and then stepping the environment with all the actions in the parent process. However, I am not sure how to implement this other than trying `multiprocessing.Pool`. I'm sorry I can't be of more help, let us know how your multi-drone simulation goes and whether you need any additional help!

closed 2025-04-28 by Federico-PizarroBejarano