learnsyslab/safe-control-gym #208: Make paper results reproducible in version 1.0.0

cblancovolle, 2025-09-10:
Currently, experiments for papers _Safe Learning in Robotics: From Learning-Based Control to Safe Reinforcement Learning_ and _Safe-Control-Gym: A Unified Benchmark Suite for Safe Learning-Based Control and Reinforcement Learning in Robotics_ are not easily reproducible in current version of the framework. 

It seems the code for generating the figures are available in pre-release [v1.0.0-alpha](https://github.com/utiasDSL/safe-control-gym/releases/tag/v1.0.0-alpha). However, I did not manage to run the scripts due to package installation errors.

adamhall, 2025-09-16:
Hi @cblancovolle,

Can you list the machine you are working on, the python environment you are working with, and the errors you are receiving so I can help you further.

We dropped support for the experiments as it became too rescritving when trying to expand the capabilities of the gym, however, they should still work in older versions of the gym. If you share the info above we can try and help you to get it working.

cblancovolle, 2025-09-17:
Hi thanks for your answers and the clarification about the support for experiments.

@Federico-PizarroBejarano @adamhall  I was successfully able to run the experiments to generate figures for paper _Safe Learning in Robotics: From Learning-Based Control to Safe Reinforcement Learning_ using release [0.5.0](https://github.com/utiasDSL/safe-control-gym/releases/tag/v0.5.0).

I would like to also run the experiments showed in _Safe-Control-Gym: A Unified Benchmark Suite for Safe Learning-Based Control and Reinforcement Learning in Robotics_. As indicated in [README.md](https://github.com/utiasDSL/safe-control-gym?tab=readme-ov-file#safe-control-gym) those experiments should be available in release [0.6.0](https://github.com/utiasDSL/safe-control-gym/releases/tag/v0.6.0). However it seems that in the `experiments` folder, no experiment has been added compared to release [0.5.0](https://github.com/utiasDSL/safe-control-gym/releases/tag/v0.5.0).

[image]

That's why I tried with release [1.0.0-alpha](https://github.com/utiasDSL/safe-control-gym/releases/tag/v1.0.0-alpha) in which I noticed the presence of a folder named `arxiv` in `experiments` that seems to corresponds to the experiments I am looking for.

[image]

When trying to install from release (on Windows 11 Entreprise) [1.0.0-alpha](https://github.com/utiasDSL/safe-control-gym/releases/tag/v1.0.0-alpha) after following the instructions of `README.md`:
```
conda create --name safe python=3.8.10
conda activate safe

python -m pip install --upgrade pip
pip install -e .
```

I get the following error:

```
Collecting poetry-core@ git+https://github.com/python-poetry/poetry-core.git@master
        Cloning https://github.com/python-poetry/poetry-core.git (to revision master) to ...\appdata\local\temp\pip-install-d87v9h1w\poetry-core_d2428c7eef8d48cfa3b6fcd69887e505
        Running command git clone --filter=blob:none --quiet https://github.com/python-poetry/poetry-core.git '...\AppData\Local\Temp\pip-install-d87v9h1w\poetry-core_d2428c7eef8d48cfa3b6fcd69887e505'
        WARNING: Did not find branch or tag 'master', assuming revision or ref.
        Running command git checkout -q master
        error: pathspec 'master' did not match any file(s) known to git
        error: subprocess-exited-with-error

        git checkout -q master did not run successfully.
        exit code: 1

        See above for output.

        note: This error originates from a subprocess, and is likely not a problem with pip.
      error: subprocess-exited-with-error

      git checkout -q master did not run successfully.
      exit code: 1
```

This error can be solved by modifying `requires = ["poetry-core @ git+https://github.com/python-poetry/poetry-core.git@master"]` into `requires = ["poetry-core @ git+https://github.com/python-poetry/poetry-core.git@main"]` in `pyproject.toml`.

 But then when running `pip install -e .` we get a new error:

```
Collecting poetry-core@ git+https://github.com/python-poetry/poetry-core.git@main
        Cloning https://github.com/python-poetry/poetry-core.git (to revision main) to ...\appdata\local\temp\pip-install-m87j90d_\poetry-core_314beddfe4e145b4ad9729614d891fe1
        Running command git clone --filter=blob:none --quiet https://github.com/python-poetry/poetry-core.git '...\AppData\Local\Temp\pip-install-m87j90d_\poetry-core_314beddfe4e145b4ad9729614d891fe1'
        Resolved https://github.com/python-poetry/poetry-core.git to commit f8c3684d3d1502ae653c1ef5abebc74e895ee8da
        Getting requirements to build wheel: started
        Getting requirements to build wheel: finished with status 'error'
        error: subprocess-exited-with-error
      
        Getting requirements to build wheel did not run successfully.
        exit code: 1
      
        [32 lines of output]
        Traceback (most recent call last):
          File "...\AppData\Local\miniconda3\envs\safe1.0.0alpha\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 389, in <module>
            main()
          File "...\AppData\Local\miniconda3\envs\safe1.0.0alpha\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 373, in main
            json_out["return_val"] = hook(**hook_input["kwargs"])
          File "...\AppData\Local\miniconda3\envs\safe1.0.0alpha\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 137, in get_requires_for_build_wheel
            backend = _build_backend()
          File "...\AppData\Local\miniconda3\envs\safe1.0.0alpha\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 70, in _build_backend
            obj = import_module(mod_path)
[... 8 lines cut ...]
          File "...\AppData\Local\Temp\pip-install-m87j90d_\poetry-core_314beddfe4e145b4ad9729614d891fe1\src\poetry\core\masonry\api.py", line 12, in <module>
            from poetry.core.factory import Factory
          File "...\AppData\Local\Temp\pip-install-m87j90d_\poetry-core_314beddfe4e145b4ad9729614d891fe1\src\poetry\core\factory.py", line 18, in <module>
            from poetry.core.packages.dependency import Dependency
          File "...\AppData\Local\Temp\pip-install-m87j90d_\poetry-core_314beddfe4e145b4ad9729614d891fe1\src\poetry\core\packages\dependency.py", line 13, in <module>
            from poetry.core.constraints.generic import parse_constraint as parse_generic_constraint
[... cut ...]
          File "...\AppData\Local\Temp\pip-install-m87j90d_\poetry-core_314beddfe4e145b4ad9729614d891fe1\src\poetry\core\constraints\version\parser.py", line 17, in <module>
            @functools.cache
        AttributeError: module 'functools' has no attribute 'cache'
        [end of output]

        note: This error originates from a subprocess, and is likely not a problem with pip.
      error: subprocess-exited-with-error

      Getting requirements to build wheel did not run successfully.
      exit code: 1
```

That currently where I am blocked during installation process.

Federico-PizarroBejarano, 2025-09-19:
I think I figured out the issues. In `pyproject.toml`, use this:

```
[tool.poetry]
name = "safe-control-gym"
version = "0.5.0"
description = "PyBullet CartPole and Quadrotor environments—with CasADi symbolic a priori dynamics—for learning-based control and RL"
authors = ["Your Name <you@example.com>"]
license = "MIT"

[tool.poetry.dependencies]
python = "^3.8.10"
matplotlib = "^3.5.1"
Pillow = "^9.0.0"
munch = "^2.5.0"
PyYAML = "^6.0.2"
imageio = "^2.14.1"
dict-deep = "^4.1.2"
scikit-optimize = "^0.9.0"
pandas = "^1.4.0"
gym = "^0.23.0"
torch = "^1.10.2"
gpytorch = "^1.6.0"
ray = "^1.9.2"
tensorboard = "^2.8.0"
casadi = "^3.5.5"
pybullet = "^3.2.0"
cvxpy = "^1.1.18"
numpy = "^1.22.1"
pytope = "^0.0.4"
Mosek = "^9.3.14"
termcolor = "^1.1.0"

[tool.poetry.dev-dependencies]

[build-system]
requires = ["poetry-core>=1.0.0"]
build-backend = "poetry.core.masonry.api"
```

This allowed me to install everything, run the tests, and recreate some of the plots, but I have not tested it extensively. Let us know if you encounter any other difficulties!